adafruit/Adafruit_CircuitPython_BNO055 #2: adafruit_bno055.py throws exception (error 19) on register write during initialization

hukuzatuna, 2017-09-08:
Environment: Metro M0 Express, CircuitPython 2.0 rc2 "latest" build as of last night's release.
Output:
Traceback (most recent call last):
  File "code.py", line 71, in <module>
  File "/lib/adafruit_bno055.py", line 111, in __init__
  File "/lib/adafruit_bno055.py", line 176, in init
  File "/lib/adafruit_bno055.py", line 117, in _write_register
  File "/lib/adafruit_bno055.py", line 117, in _write_register
  File "libraries/helpers/bus_device/adafruit_bus_device/i2c_device.py", line 94, in write
OSError: 19
 
(If this actually belongs in the bundled library issues, close it and let me know and I'll open an issue there. Thanks! -Phil)

tannewt, 2017-11-16:
Is this still an issue?

hukuzatuna, 2017-11-16:
I'm not sure. I'm at work and don't have my device. I will check this evening!

tannewt, 2018-01-09:
I just tested on CircuitPython close to 2.2 and it worked fine.

caternuson, 2018-04-10:
Seems to be back:
```python
Adafruit CircuitPython 2.2.4 on 2018-03-07; Adafruit Feather M0 Express with samd21g18
>>> import board
>>> import busio
>>> import adafruit_bno055
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> sensor = adafruit_bno055.BNO055(i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_bno055.py", line 127, in __init__
  File "adafruit_bno055.py", line 138, in _write_register
  File "adafruit_bno055.py", line 138, in _write_register
  File "adafruit_bus_device/i2c_device.py", line 102, in write
OSError: 19
>>>
```

caternuson, 2018-04-12:
This seems to be related to how the wait-for-reset is being done in the driver code. From that BNO055 datasheet, there's a POR time, from reset to config mode, of 650 ms typ. (thanks to jepler in discord for pointing this out). This isn't being respected in the code which checks for chip id as a way to determine if reset is complete. For some reason, that check returns way before 650 ms and the subsequent i2c comms fail and cause the above traceback.

tannewt, 2018-04-12:
Great find! Care to make a PR?

caternuson, 2018-04-12:
I will if @deshipu doesn't want to. Probably various ways it could be fixed. He might be more familiar with quirks and trade offs with this sensor.

caternuson, 2018-04-16:
Hopefully fixed by #10

closed 2018-04-16 by caternuson