adafruit/Adafruit_CircuitPython_BNO055 #9: RuntimeError: bad chip id (1 != a0) on ESP8266

caternuson, 2018-04-10:
Not sure if this is related to #2 or not. The traces look different, so maybe **bitbangio** is playing into it?
```python
Adafruit CircuitPython 2.2.4 on 2018-03-07; ESP module with ESP8266
>>> import board
>>> import busio
>>> import adafruit_bno055
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> sensor = adafruit_bno055.BNO055(i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_bno055.py", line 125, in __init__
RuntimeError: bad chip id (1 != a0)
>>> 
```
[image]

tannewt, 2018-04-11:
Could you zoom in on the trace so we can see what the response was? Thanks!

caternuson, 2018-04-12:
Here's a zoom on the response. Also attached the trace file.

[image]

[esp8266_bno55.zip](https://github.com/adafruit/Adafruit_CircuitPython_BNO055/files/1903680/esp8266_bno55.zip)

caternuson, 2018-04-13:
Hopefully fixed by #10

caternuson, 2018-04-25:
I'm thinking this has something to do with bitbangio and clock stretching.

I didn't even need to use the drivers. I was able to see an issue with just this simple test:
```python
>>> import board, busio
>>> from adafruit_bus_device.i2c_device import I2CDevice
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> device = I2CDevice(i2c, 0x28)
```
Tested with a Feather M0 Express and Feather ESP8266 using a TCS347245 and a BNO055. Here's what it looks like under the scope:

Feather M0 Express w/ TCS34725
No clock stretching, all good:
[image]

Feather M0 Express w/ BNO055
Some clock stretching in the ACK, but it handles it, all good:
[image]

Feather ESP8266 w/ TCS34725
No clock stretching, all good:
[image]

Feather ESP8266 w/ BNO055
It doesn't like that clock stretch :(
[image]

deshipu, 2018-04-26:
The code for clock stretching in bitbangio is here: https://github.com/adafruit/circuitpython/blob/master/extmod/machine_i2c.c#L56-L61

Personally I don't see anything wrong with it, but maybe I'm missing something...

deshipu, 2018-04-26:
Hmm, looking at the `shared-module`,  it seems the code I linked above is not used. Instead, this is used: https://github.com/adafruit/circuitpython/blob/master/shared-module/bitbangio/I2C.c#L53-L55 which is the old version with the time stretch timeout hard-coded to 255µs — but that should still work, as on the trace the stretch is a bit over 100µs, so I still have no idea.

deshipu, 2018-04-26:
Could it be that there is enough delay between releasing the SCL pin and releasing SDA pin, that ACK is missed?

caternuson, 2018-05-24:
Blaming this on clock stretching may be a red herring.

It seems to be more related to specific timing issues with specific sensors. As an example, can get the same failure on the Feather M0 Express w/ BNO055 combo using `bitbangio` instead of `busio`, like this:
```python
>>> import board, bitbangio
>>> from adafruit_bus_device.i2c_device import I2CDevice
>>> i2c = bitbangio.I2C(board.SCL, board.SDA)
>>> device = I2CDevice(i2c, 0x28)
```
and here's the trace:
[image]

kattni, 2020-05-04:
@caternuson Is this still an issue?

caternuson, 2020-05-05:
@kattni Unknown. The landscape has changed so much (ESP8266 is no longer supported and bitbangio isn't in M0 builds) that I'd say let's just close this and if there is still some issue, we can open a new one with updated info and testing when we hit it.

closed 2020-05-05 by ladyada