adafruit/Adafruit_CircuitPython_BNO055 #17: calibrated and calibration_status do not seem to be working

ViennaMike, 2018-11-09:
I can't see why they won't, but when I try to get either the calibrated or calibration_status attributes, I get a not found error. Here's my test code and result:

```
import time
import board
import busio
import adafruit_bno055

i2c = busio.I2C(board.SCL, board.SDA)
sensor = adafruit_bno055.BNO055(i2c)
while True:
    print('Temperature: {} degrees C'.format(sensor.temperature))
    print('Accelerometer (m/s^2): {}'.format(sensor.accelerometer))
    print('Magnetometer (microteslas): {}'.format(sensor.magnetometer))
    print('Gyroscope (deg/sec): {}'.format(sensor.gyroscope))
    print('Euler angle: {}'.format(sensor.euler))
    print('Quaternion: {}'.format(sensor.quaternion))
    print('Linear acceleration (m/s^2): {}'.format(sensor.linear_acceleration))
    print('Gravity (m/s^2): {}'.format(sensor.gravity))
    print()
#    print('Calibrated: {}'.format(sensor.calibrated))
    print('Calibration Status: {}'.format(sensor.calibration_status))
    time.sleep(1)
```

Temperature: 128 degrees C
Accelerometer (m/s^2): (0.0, 0.0, 0.0)
Magnetometer (microteslas): (0.0, 0.0, 0.0)
Gyroscope (deg/sec): (-0.003333333333333333, -0.0022222222222222222, -0.01)
Euler angle: (0.0, 0.0, 0.0)
Quaternion: (0.0, 0.0, 0.0, 0.0)
Linear acceleration (m/s^2): (0.0, 0.0, 0.0)
Gravity (m/s^2): (0.0, 0.0, 0.0)

Traceback (most recent call last):
  File "imutest.py", line 19, in <module>
    print('Calibration Status: {}'.format(sensor.calibration_status))
AttributeError: 'BNO055' object has no attribute 'calibration_status'

ViennaMike, 2018-11-10:
Problem was the calibration status wasn't working. Turns out, it's because the version in Pypi is behind and doesn't have the latest changes that are in gitHub.

kattni, 2018-11-10:
Sounds like you figured it out. I'll get another release out to update the version on PyPI. Thanks for catching this!

closed 2018-11-10 by kattni