adafruit/Adafruit_CircuitPython_BNO055 #26: mode setter appears wrong

peterhinch, 2019-05-03:
I'm writing a MicroPython driver for your board. In my testing it is only possible to change the device mode if you first change the mode to `_CONFIG_MODE`. This does not seem to be documented in the datasheet. Unfortunately I don't have any hardware capable of running CircuitPython but I think the following test will fail:
```python
imu.mode = 5
print(imu.mode)  # I think this will be unchanged
```
Before I submit a PR I would be grateful if someone could verify if this is the case.

For reference this is my mode setting code. This includes delays mandated by the device datasheet.
```python
    def mode(self, new_mode=None):
        old_mode = self._read(_MODE_REGISTER)
        if new_mode is not None:
            self._write(_MODE_REGISTER, _CONFIG_MODE)  # This is empirically necessary if the mode is to be changed
            time.sleep_ms(20)  # Datasheet table 3.6
            if new_mode != _CONFIG_MODE:
                self._write(_MODE_REGISTER, new_mode)
                time.sleep_ms(10)  # Table 3.6
        return old_mode
```

caternuson, 2019-05-03:
Hmmm...I think you are correct. 

```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit ItsyBitsy M4 Express with samd51g19
>>> import adafruit_bno055
>>> from busio import I2C
>>> from board import SDA, SCL
>>> i2c = I2C(SCL, SDA)
>>> imu = adafruit_bno055.BNO055(i2c)
>>> imu.mode
12
>>> imu.mode = 5
>>> imu.mode
12
>>> imu.mode = 0
>>> imu.mode
0
>>> imu.mode = 5
>>> imu.mode
5
>>>
```

peterhinch, 2019-05-04:
Thank you for confirming. I'll submit a PR in a day or two.

caternuson, 2019-05-09:
Fixed by #27 

Thanks @peterhinch!

closed 2019-05-09 by caternuson

tannewt, 2019-05-09:
@peterhinch email me (scott@adafruit.com) if you'd like some hardware to test CircuitPython with. We really appreciate your fixes to the drivers.