adafruit/Adafruit_CircuitPython_BNO055 #28: The reset method seems inconsistent

peterhinch, 2019-05-06:
The BNO055 class **reset** method is user-accesible https://github.com/adafruit/Adafruit_CircuitPython_BNO055/blob/32a5a8a20b8a5d00d20cc5370271c36a0506791b/adafruit_bno055.py#L148-L156
If a user calls this method, my reading of the datasheet is that the behaviour of the chip is equivalent to a power-on or a pulse on the reset line. It therefore seems inconsistent that the constructor, after calling reset(), issues
https://github.com/adafruit/Adafruit_CircuitPython_BNO055/blob/32a5a8a20b8a5d00d20cc5370271c36a0506791b/adafruit_bno055.py#L128-L133
Either these lines are redundant or they should be included in the **reset** method. Alternatively user access could be discouraged by renaming it **_reset**.

tannewt, 2019-05-10:
My preference would be to rename it to `_reset` and reduce the API surface area.

dherrada, 2019-10-16:
Fixed in #31

closed 2019-10-16 by dherrada