adafruit/Adafruit_CircuitPython_BNO055 #33: sensor.gyroscope RPS not DPS

CodeMaterial, 2019-08-15:
The old Python BNO055 library has an issue where the gyroscope method returns RPS not DPS as defined in the documentation.

The UNIT_SEL address hasn't been set to degrees and there's no conversion to degrees in either the new or old library.

I'm still using all the old Python libraries as I use boards that don't support Circuit-Python currently. It's a shame they've been archived as I got stuck on this issue for days.

Thanks,

Sam.

caternuson, 2019-08-15:
Are you opening an issue for this repo or the older archived repo? It's not clear.

CodeMaterial, 2019-08-15:
The issue definitely exist in the old repository and it looks like it exists in this one as well. I would test it but circuit python doesn't support my hardware. (Jetson TX2)

makermelissa, 2019-08-15:
Are you sure the Jetson TX2 isn't supported? I thought it was added back in version 1.25 of the Blinka library.

CodeMaterial, 2019-08-15:
I'll re-test it tomorrow as i can't remember fully what went wrong but from my memory it was a "Jetson.GPIO not found" error.

If it works I'll see if sensor.gyroscope returns radians or degrees too.

makermelissa, 2019-08-15:
If you do get an error like that, please submit an issue to https://github.com/adafruit/Adafruit_Blinka/issues as I don't think there is one already specifically for the TX2.

ladyada, 2019-08-15:
ok closing this cause its not an issue that goes here :) plz open a new issue if this library doesnt work

closed 2019-08-15 by ladyada

caternuson, 2019-08-15:
DPS is the default:
[image]
and I'm not seeing anything in the driver that is changing that.

Are you saying you are getting RPS?

CodeMaterial, 2019-08-16:
Hi @caternuson @ladyada 

I've tried to install the CircuitPython version of the BNO055 library on my Nvidia Jetson TX2 to no avail. However this is a completely different issue to the DPS vs RPS issue so will open a separate ticket if I exhaust all other options. All comments below relate to the Adafruit_Python_BNO055 library but might also apply to Adafruit_CircuitPython_BNO055.

The attached image shows a graph of the library's gyro measurements when rotating the sensor around the x axis around 1RPS for 3 seconds then back again.

This is recorded over ROS but a similar issue can be seen when just printing out the gyro measurements from the simpletest.py script with the only change being `bno = BNO055.BNO055(busnum=0)` and `print(bno.read_gyroscope())` (output seen below)

I've read through the datasheet and agree with @caternuson that it should be in DPS. The only thing I can think of is that mine has been set to RPS and has some form of persistent memory which I don't think this chip has. All very odd.

I have tried to clean install the BNO055 library and use as close to the examples as possible but  please do test it your end as it might just be something weird with my setup.

Hope this helps,

Sam.

[image]

```
Heading=323.69 Roll=-32.56 Pitch=6.44	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.3111111111111111, -2.2977777777777777, -0.21888888888888888)
Heading=323.25 Roll=-21.19 Pitch=7.81	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.25, -1.8144444444444445, 0.0044444444444444444)
Heading=322.88 Roll=-10.25 Pitch=9.12	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.2911111111111111, -3.2866666666666666, -0.03111111111111111)
Heading=322.88 Roll=2.19 Pitch=10.44	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.08555555555555555, -1.1855555555555555, -0.10555555555555556)
Heading=322.81 Roll=11.12 Pitch=11.31	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.1988888888888889, -1.3266666666666667, -0.18)
Heading=322.31 Roll=21.75 Pitch=13.25	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.39666666666666667, -2.421111111111111, -0.17888888888888888)
Heading=322.12 Roll=33.56 Pitch=16.56	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.3977777777777778, -2.298888888888889, -0.18555555555555556)
Heading=321.88 Roll=42.19 Pitch=19.75	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.3422222222222222, -2.968888888888889, -0.07888888888888888)
Heading=322.50 Roll=55.62 Pitch=26.94	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.46555555555555556, -2.3722222222222222, -0.35888888888888887)
Heading=323.94 Roll=65.19 Pitch=34.50	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.1922222222222222, -1.8355555555555556, -0.1988888888888889)
Heading=323.88 Roll=73.69 Pitch=54.50	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.2733333333333333, -2.3733333333333335, -0.12222222222222222)
Heading=323.88 Roll=77.38 Pitch=96.31	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.18555555555555556, -1.3944444444444444, -0.03666666666666667)
Heading=322.88 Roll=75.62 Pitch=114.75	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.043333333333333335, -0.07333333333333333, 0.18)
Heading=322.56 Roll=76.19 Pitch=90.94	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.3, 3.187777777777778, -0.10555555555555556)
Heading=324.75 Roll=63.69 Pitch=31.06	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.5611111111111111, 3.611111111111111, -0.16444444444444445)
Heading=325.81 Roll=50.00 Pitch=18.88	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.22777777777777777, 2.2222222222222223, 0.056666666666666664)
Heading=326.75 Roll=39.06 Pitch=14.75	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.1688888888888889, 2.201111111111111, 0.013333333333333334)
Heading=327.88 Roll=25.12 Pitch=11.56	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.24444444444444444, 2.7733333333333334, -0.0077777777777777776)
Heading=328.31 Roll=16.00 Pitch=10.31	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.024444444444444446, 1.4322222222222223, -0.07555555555555556)
Heading=329.25 Roll=6.06 Pitch=9.62	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.08888888888888889, 1.9055555555555554, -0.035555555555555556)
Heading=330.12 Roll=-4.50 Pitch=8.88	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.12666666666666668, 1.8677777777777778, -0.0011111111111111111)
Heading=330.69 Roll=-14.19 Pitch=8.44	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(0.09666666666666666, 1.7933333333333332, 0.03888888888888889)
Heading=331.31 Roll=-23.94 Pitch=8.62	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.0077777777777777776, 1.8888888888888888, 0.07222222222222222)
Heading=332.06 Roll=-36.75 Pitch=9.94	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.0811111111111111, 2.5077777777777777, 0.02666666666666667)
Heading=332.38 Roll=-47.31 Pitch=12.25	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.15777777777777777, 2.311111111111111, 0.0022222222222222222)
Heading=331.44 Roll=-58.75 Pitch=16.75	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.39555555555555555, 2.6277777777777778, 0.07666666666666666)
Heading=330.19 Roll=-73.94 Pitch=32.19	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.35555555555555557, 1.1633333333333333, 0.2288888888888889)
Heading=328.50 Roll=-76.94 Pitch=40.88	Sys_cal=0 Gyro_cal=3 Accel_cal=0 Mag_cal=1
(-0.35555555555555557, 0.26222222222222225, 0.02666666666666667)
```

ladyada, 2019-08-16:
hiya we wont be doing any code changes or support for the old library - its archived for a reason :) lets get your nvida working, that way we only look at this code