adafruit/Adafruit_CircuitPython_BNO055 #34: Update gyroscope to match design guide

caternuson, 2019-08-15:
This #33 made me check the Design Guide on [sensor properties and units](https://circuitpython.readthedocs.io/en/latest/docs/design_guide.html#sensor-properties-and-units).

`gyroscope` should be `gyro` and the units should be radians per second.

This could be shimmed in like `acceleration` and `magnetic` appear to have been. So added in, not replaced:
```python
    gyroscope = _ScaledReadOnlyStruct(0x14, '<hhh', 1/16)
    """Gives the raw gyroscope reading in degrees per second."""
    gyro = _ScaledReadOnlyStruct(0x14, '<hhh', 1/16) * 0.017453292519943295
    """Gives the raw gyroscope reading in radians per second."""
```
or just bury it all in the scale:
```python
    gyro = _ScaledReadOnlyStruct(0x14, '<hhh', 0.001090830782496456)
    """Gives the raw gyroscope reading in radians per second."""
```
or be all fancy and use `math.pi`:
```python
    gyro = _ScaledReadOnlyStruct(0x14, '<hhh', math.pi / 2880)
    """Gives the raw gyroscope reading in radians per second."""
```

The other approach would be to actually change the units via `UNIT_SEL`, but that would require more code changes.

ladyada, 2019-08-15:
im cool with mathing it :)

caternuson, 2019-09-09:
mathed it in #35

closed 2019-09-09 by caternuson