adafruit/Adafruit_CircuitPython_BNO055 #41: Add UART functionality to library

kattni, 2019-10-25:
The [original Python library](https://github.com/adafruit/Adafruit_Python_BNO055/blob/master/Adafruit_BNO055/BNO055.py) includes the ability to use UART with the BNO055. Please look into implementing UART in the CircuitPython library.

ladyada, 2019-10-25:
@dherrada hiya you recently did some hacking on this library so thought this could be a good one for you to do. the sensor has both i2c and uart. the old library had both, this library only has i2c but maybe we could add uart?

caternuson, 2019-10-25:
Looks like this will be interesting from a class layout perspective. Current library only has the one class for I2C. So did the old library, and determined I2C vs. UART based on what was used in constructor. That's different than most of the CP libraries that tend to have separate classes for each interface type.

ladyada, 2019-10-25:
im ok with a refactor to have two classes!

caternuson, 2019-10-25:
Cool. Refactor fun time! @dherrada Let me know if you're on this.

dherrada, 2019-10-28:
@caternuson Yep. I'll do it. I probably won't be able to get to it until after Nov. 1 (early decision deadline for V. Tech) but I'm definitely interested in learning about this and adding uart

caternuson, 2019-10-28:
@dherrada awesome. go for it.

tannewt, 2019-11-01:
Could we abstract UART, I2C and SPI away instead and have the driver deal with commands? I'd rather see something like the display buses where the driver doesn't care about the transport rather than having multiple classes.

ladyada, 2019-11-01:
yes we would abstract it like we do for BMx80 sensors or PN532 for an exciting look https://github.com/adafruit/Adafruit_CircuitPython_PN532/tree/master/adafruit_pn532

tannewt, 2019-11-01:
I'm thinking we'd have one driver class that takes in a standard wrapper for UART, I2C and SPI. I don't mean having a number of subclasses. I imagine there is a standard way to pack things over the different transports and by having an object to pass in, we can factor out the packing logic.

ladyada, 2019-11-02:
for i2c/spi you can do that kinda cause SPI chips tend to use a standard way to mimic i2c registers. 
see
https://github.com/adafruit/Adafruit_BusIO/blob/master/examples/i2corspi_register/i2corspi_register.ino
uart tends to be a free-for-all, as its quite rare. we could try all the sensors with uart and then see if its generalizable

kattni, 2020-04-16:
@caternuson If you have cycles for this, it's something that still needs to be done. Dylan is currently on other projects.

caternuson, 2020-04-24:
Done with #47

closed 2020-04-24 by caternuson