adafruit/Adafruit_CircuitPython_BNO055 #48: UART mode only allows for 10Hz data reads

blakerutledge, 2020-04-29:
If I modify the lib manually and decrease the sleep from 0.1 to 0.01 in the UART mode, I can read the data at 100Hz just fine.

Is there a reason these sleep values are so high and throttling the data?

dherrada, 2020-06-25:
@blakerutledge I'm not sure. Have you encountered any issues using it at 100Hz? If not, then make a PR and I can test (or I can make the PR if you'd prefer that)

blakerutledge, 2020-06-26:
Hi @dherrada! I am on a different machine for work now, I don't have the repo pulled for a PR, so you should handle. It was just those two lines where sleep in UART was 0.1seconds instead of 0.01. 
I didn't have any issues! Curious to see your tests

dherrada, 2020-06-26:
@blakerutledge Ok, that's fine. Yeah, it didn't seem like there were too many changes, but I just wanted to make sure that changing the time.sleep was all you did. Thanks

dherrada, 2020-07-10:
Fixed by #50, which improves stability and speed