adafruit/Adafruit_CircuitPython_BNO055 #55: Segmentation Fault

calvaradocl, 2020-07-31:
Hi

We are experiencing sometimes a segmentation fault when trying to run the example code of the BNO055 unit, is there a fix for this? We got our BNO055 to run, but after some time it stopped working and we are receiving the Segmentation Fault message, tried to debug it externally by running python inside a debugger without results.

dherrada, 2020-08-06:
Hey, do you mind posting the code you're using and the output you're getting?

calvaradocl, 2020-08-06:
Of course, I'm using the sample code of the adafruit site:

import time
import board
import busio
import adafruit_bno055
 
# Use these lines for I2C
i2c = busio.I2C(board.SCL, board.SDA)
sensor = adafruit_bno055.BNO055_I2C(i2c)
 
# User these lines for UART
# uart = busio.UART(board.TX, board.RX)
# sensor = adafruit_bno055.BNO055_UART(uart)
 
while True:
    print("Temperature: {} degrees C".format(sensor.temperature))
    print("Accelerometer (m/s^2): {}".format(sensor.acceleration))
    print("Magnetometer (microteslas): {}".format(sensor.magnetic))
    print("Gyroscope (rad/sec): {}".format(sensor.gyro))
    print("Euler angle: {}".format(sensor.euler))
    print("Quaternion: {}".format(sensor.quaternion))
    print("Linear acceleration (m/s^2): {}".format(sensor.linear_acceleration))
    print("Gravity (m/s^2): {}".format(sensor.gravity))
    print()
 
    time.sleep(1)

Yesterday I updated my RPI 3B+ to the latest firmware and the problem was gone (I upgraded my system from 4.19.118 to 5.4.51 by using apt-get upgrade).

I don't know if there is a bug working with 4.19.118, as it was working without problems but after rebooting my Pi the issues ocurred, even after I made a fresh install (the problem ocurred again). Only upgrading to the latest firmware resolved the issue.

ladyada, 2020-08-06:
the BNO055 requires clock stretching support, there have been firmware issues with clock stretching so not surprising

@dherrada please add a warning/note to the guide!

closed 2020-08-06 by ladyada

dherrada, 2020-08-06:
@ladyada, will do.

calvaradocl, 2020-08-06:
Yes, please let me add the following, after reviewing some technical reviews on the web, I found out (and validated) that working with the parameter i2c_arm_baudrate on 200000 behaves better than making it smaller.

Please take a look at the following URL: https://softsolder.com/2018/09/04/raspberry-pi-i2c-bus-timing-vs-bno055-clock-stretching/.

I made some tests, and after setting the Pi at that baud rate, and letting the BNO055 work for three days straight, no wrong measurements so far.