adafruit/Adafruit_CircuitPython_BNO055 #56: Micropython requirement

TurryBoeing, 2020-09-14:
Hello, it seems that Micropython is required, but I don´t know how to installed, since it seems that it is not available on pip anymore, or I can´t find it:

Traceback (most recent call last):
  File "python3gyrotests.py", line 4, in <module>
    from elements.ada_bno055 import *
  File "/home/pi/turrydronetestspython3/elements/ada_bno055.py", line 37, in <module>
    from micropython import const

Best regards.

ladyada, 2020-09-14:
did you install blinka according to https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

TurryBoeing, 2020-09-14:
Dependency is already met.

`pi@bpi-iot-ros-ai:~/turrydronetestspython3 $ sudo pip3 install adafruit-blinka
Requirement already satisfied: adafruit-blinka in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: sysv-ipc; platform_system != "Windows" and platform_machine != "mips" in /usr/local/lib/python3.5/dist-packages (from adafruit-blinka)
Requirement already satisfied: Adafruit-PlatformDetect>=2.11.1 in /usr/local/lib/python3.5/dist-packages (from adafruit-blinka)
Requirement already satisfied: RPi.GPIO in /usr/local/lib/python3.5/dist-packages (from adafruit-blinka)
Requirement already satisfied: Adafruit-PureIO>=1.1.5 in /usr/local/lib/python3.5/dist-packages (from adafruit-blinka)
Requirement already satisfied: rpi-ws281x>=4.0.0 in /usr/local/lib/python3.5/dist-packages (from adafruit-blinka)
Requirement already satisfied: pyftdi>=0.40.0 in /usr/local/lib/python3.5/dist-packages (from adafruit-blinka)
Requirement already satisfied: pyusb>=1.0.0 in /usr/local/lib/python3.5/dist-packages (from pyftdi>=0.40.0->adafruit-blinka)
Requirement already satisfied: aenum>=2.1.0; python_version == "3.5" in /usr/local/lib/python3.5/dist-packages (from pyftdi>=0.40.0->adafruit-blinka)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi>=0.40.0->adafruit-blinka)
pi@bpi-iot-ros-ai:~/turrydronetestspython3 $
pi@bpi-iot-ros-ai:~/turrydronetestspython3 $
pi@bpi-iot-ros-ai:~/turrydronetestspython3 $
pi@bpi-iot-ros-ai:~/turrydronetestspython3 $ uname -a
Linux bpi-iot-ros-ai 4.4.55-BPI-M2Z-Kernel #1 SMP PREEMPT Fri Apr 10 16:05:55 HKT 2020 armv7l GNU/Linux`

Could the issue be related with the fact that I am trying this on a Banana Pi? (See output of uname -a)

ladyada, 2020-09-14:
oh yeah, we dont support banana pi

tannewt, 2020-09-14:
Here is a tutorial on adding support for a single board computer to Blinka: https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka

Closing this issue since it isn't an issue with the BNO055 library.

closed 2020-09-14 by tannewt