adafruit/Adafruit_CircuitPython_BNO055 #62: Library too large for smaller boards

caternuson, 2020-11-09:
```python
Adafruit CircuitPython 6.0.0-rc.1 on 2020-11-03; Adafruit QT Py M0 with samd21e18
>>> import adafruit_bno055
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_bno055.py", line 111, in <module>
MemoryError: memory allocation failed, allocating 776 bytes
>>> 
```

ladyada, 2020-11-10:
yeah not surprising, its a complex library

ATMakersBill, 2020-11-10:
Carter was good enough to play with me on this tonight (as was Dan earlier).  Goal was to make a zero-solder Head Mouse using just the QTPy and the BNO055 Stemma QT board.  We got it working.

I built a custom .UF2 with HID, BusDevice, and Register Frozen in and lots of features disabled.  Here's the relevant part of mpconfigboard.mk:
`CIRCUITPY_ANALOGIO = 0
CIRCUITPY_PULSEIO = 0
CIRCUITPY_PWMIO = 0
CIRCUITPY_MATH = 0
#CIRCUITPY_NEOPIXEL_WRITE = 0
CIRCUITPY_ROTARYIO = 0
CIRCUITPY_RTC = 0
CIRCUITPY_SAMD = 0
CIRCUITPY_USB_MIDI = 0
CIRCUITPY_TOUCHIO = 0
CIRCUITPY_TOUCHIO_USE_NATIVE = 0
CIRCUITPY_FULL_BUILD = 0

SUPEROPT_GC = 0

FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_HID
FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_BusDevice
FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_Register`

With that in place, we still didn't have enough RAM for the library so Carter pulled out the custom Struct classes that were wrapping the values.  The BNO055_I2C class (in this hacked version) worked just like the _UART version and read the registered, formatted them and sent them.

It worked well, but still won't fit in the stock .UF2 as built for the QTPy.

For my (short term demo) purposes, this is a fine hack, and I love that I have a head mouse I can show in my presentation tomorrow (with no soldering!).

However, I do have two thoughts:

1. Whether it's a forked "lite" version of the library or not, it would be nice to have a version that would run on the M0 without external RAM.  Between the Trinket & especially the QTPy, we make lots of "Whatever to HID" custom devices and "9DOF->HID" is a huge addition.  I don't THINK it would be easier to use one of the other 9DOFs and run the Fusion code in the M0, so pruning this library to fit seems worth the effort.
2. With the QTPy shipping without the extra RAM, perhaps the build for it should be tuned a bit.  Add in BusDevice, HID as frozen and cut out pulsio and some others (maybe not all I did).  But given that there is a separate build for the QTPy + RAM, perhaps that's the one that gets all the features, and there's one that has a bit of extra room.  I do think that HID + I2C is a fantastic use case for the $6 device and can't wait to do more with it.

I'll continue to work with Carter to see where this ends up - just wanted to add to the issue & say thanks.
Bill

caternuson, 2020-11-10:
fork/branch ref FWIW:
https://github.com/caternuson/Adafruit_CircuitPython_BNO055/tree/qtpy_hack

nothing fancy. just reverted to basic i2c xfers instead of using register classes.

ATMakersBill, 2020-11-10:
You also commented out all those constant declarations (which @dhalbert reminded me take a huge amount of memory for what they are).  Perhaps those could move into a separate class like HID does.

wikematkins, 2021-02-05:
Just adding that I ran into this issue as well. Seems like the issue could just as well be "board too small for larger libraries".