adafruit/Adafruit_CircuitPython_BNO055 #63: How to do calibration?

maha-haji, 2020-11-14:
Hi,

Does the device need calibration? If so, how can this be done? The online tutorial only mentions device calibration for the use with Arduino.

Thanks!

ladyada, 2020-11-15:
it self-calibrates, you shouldnt need to do anything

maha-haji, 2020-11-17:
I see. No matter how long I leave it on for, however, it still says calibration for most sensors is zero (Sys, Accel, and Mag). Only Gyro is at a value of 3. Is there an additional step I need to include in my code? Thanks!

ladyada, 2020-11-17:
you can check the datasheet for more details, probably spin it around a bit like you dowith your phone!

closed 2020-11-17 by ladyada

maha-haji, 2020-11-17:
Is there way to save the calibration once it’s done? I need to leave this sensor in place and need it to restart with a system on it’s own. I was able to do that when I was using an Arduino, but don’t see that option in CircuitPython hookup.

ladyada, 2020-11-18:
no it must continuously calibrate