adafruit/Adafruit_CircuitPython_BNO055 #65: Initialisation fails due to reset in __init__.

r0the, 2020-12-09:
While executing bno055_simpletest.py, the initialisation of the BNO055 always fails with the
following error:

```
Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "/lib/adafruit_bno055.py", line 671, in __init__
  File "/lib/adafruit_bno055.py", line 185, in __init__
  File "/lib/adafruit_bno055.py", line 677, in _write_register
  File "/lib/adafruit_bno055.py", line 677, in _write_register
  File "adafruit_bus_device/i2c_device.py", line 102, in write
OSError: [Errno 5] Input/output error
```

The error occurs in the first I2C write after the reset in `BNO055.__init__`.

I'm currently porting an Arduino-based application to CircuitPython. On Arduino, the same BNO works fine with the
corresponding Adafruit library. I made some comparisons of the two libraries and noticed that the Arduino library
doesn't reset the BNO. So I removed the `self._reset()` line in and now the bno055_simpletest.py works fine.

I'm not experienced enough with CircuitPython yet to see what's causing this behaviour.

ladyada, 2020-12-09:
what board are you using

siddacious, 2020-12-09:
@r0the please include more information about what devices you're using to test.

The most likely cause is that the sensor isn't yet fully operational after the reset, however the sleep in the reset function tracks with what the datasheet suggests, so it _shouldn't_ be a problem. That said, you may wish to try increasing the sleep duration here:
https://github.com/adafruit/Adafruit_CircuitPython_BNO055/blob/master/adafruit_bno055.py#L203

Please let us know if that helps, along with any other information you can provide about how the CircuitPython code is being run.

r0the, 2020-12-09:
Thanks for your quick response.

@ladyada I'm using a Feather M4 Express connected to a custom-built sensor board containing the BNO055 and several other I2C-based sensors. The BNO board is not an Adafruit product, it's a [Pesky BNO055 Nano Board](https://www.tindie.com/products/onehorse/wearable-bno055-nano-board/).

@siddacious I increased the sleep duration up to 5 seconds, the initialisation still fails the same way.

Some additional information:
- CircuitPython 6.0.0
- Script is run from Thonny 3.3.1 on macOS

It's no big deal, since I can get it working by commenting out the reset(), it just bugs me.

r0the, 2020-12-11:
I did some digging and I think that what I'm experiencing is related to [this issue](https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BNO055-power-on-reset-issues/td-p/6192):

> We recently discovered a bug in the reset process when the BNO055 expects a crystal to be connected. This means that the BNO055 will fail to reset if there is no external crystal attached.
> 
> To prevent the issue:
> 1. If possible, use an external crystal;
> 2. If 1 is not possible, avoid triggering a reset after the part has been turned on.
>
> Current workaround:
>
> Until a new firmware is provided, to reset the BNO055 to a known state, write every configuration register to a known value of your choice, such as the default value (or other).

Would you please consider to add an option to omit the reset in order to support the workaround suggested by Bosch for sensors without an external crystal?

ladyada, 2020-12-11:
sounds good - please ask pesky to submit a PR, we cannot test such hardware because we odnt own that PCB

jposada202020, 2021-04-24:
@r0the Hello :) any luck getting the PR from pesky? let me know thanks

caternuson, 2022-11-04:
Closing. Can't reproduce using Adafruit BNO055 breakout with a Feather M4:
```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather M4 Express with samd51j19
>>> import bno055_simpletest
Temperature: 0 degrees C
Accelerometer (m/s^2): (0.0, 0.0, 0.0)
Magnetometer (microteslas): (17.25, -30.25, -61.375)
Gyroscope (rad/sec): (-0.00218166, -0.00327249, 0.00218166)
Euler angle: (0.0, 0.3125, 3.125)
Quaternion: (0.999634, -0.0275269, -0.00280762, 0.0)
Linear acceleration (m/s^2): (0.0, 0.0, 0.16)
Gravity (m/s^2): (0.05, -0.53, 9.79)

Temperature: 23 degrees C
Accelerometer (m/s^2): (0.03, -0.54, 9.97)
Magnetometer (microteslas): (17.25, -29.875, -59.875)
Gyroscope (rad/sec): (-0.00109083, -0.00109083, 0.0)
Euler angle: (0.0, 0.3125, 3.125)
Quaternion: (0.999634, -0.0275269, -0.00286865, 0.0)
Linear acceleration (m/s^2): (0.0, 0.0, 0.16)
Gravity (m/s^2): (0.05, -0.53, 9.79)
```