adafruit/Adafruit_CircuitPython_BNO055 #66: Re-using calibration data

RicDG, 2020-12-11:
As a newcomer to this great sensor and python, I am getting good data using bno055_simpletest.py. It would be very useful if we had the option to save the self-calibration data for re-use at start-up, where repeated calibration is unecessary or causes delays. My searches on the internet indicate that other BNO055 users have the same issue, but have I have not found any workable paython code.

ladyada, 2020-12-12:
is there any code for doing so on other platforms?

RicDG, 2020-12-13:
I found https://github.com/adafruit/Adafruit_BNO055/blob/master/examples/restore_offsets/restore_offsets.ino
I also found a recommended 10 step process summarised by a Bosch community moderator at https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/bno055-related-questions/m-p/16432
I am using the BNO055 with a Pi Zero and Python.

ladyada, 2020-12-16:
ok thanks, we have no ETA for when that would be added so leaving this open. if someone can submit a PR for it, that would be great!

FoamyGuy, 2024-11-26:
This was resolved by #119, there is now an example that shows how to do the calibration.

closed 2024-11-26 by FoamyGuy