adafruit/Adafruit_CircuitPython_BNO055 #67: Raspberry Pi erroneous data

ender18g, 2021-01-01:
Thanks for the library.  I've use the bno055_i2c-gpio_simpletest.py in the example folder to read sensor data from a bno055 connected directly to a Raspberry pi 4.  I've tried bus 0 for the i2c connection, but I'm still getting erroneous data. much of the data is filled with "None" values and the temperature will usually read below -100 C.

Is there a specific bus that I should be using? I know that the clock stretching issue is a challenge, but I hoped that this example file had circumvented that issue. I have followed the /boot/config.txt instructions in the sample code comments. Thanks again for any advice.

Output below:

Temperature: -102 degrees C
Accelerometer (m/s^2): (-0.31, -0.3, 9.42)
Magnetometer (microteslas): (-23.5, -30.0625, -16.875)
Gyroscope (rad/sec): (0.002181661564992912, 0.003272492347489368, 0.002181661564992912)
Euler angle: (None, None, None)
Quaternion: (0.0, 0.0, 0.0, 0.0)
Linear acceleration (m/s^2): (0.0, 0.0, 0.0)
Gravity (m/s^2): (None, None, None)

Temperature: -102 degrees C
Accelerometer (m/s^2): (-0.27, -0.3, 9.43)
Magnetometer (microteslas): (-22.0625, -28.25, -14.875)
Gyroscope (rad/sec): (0.001090830782496456, 0.001090830782496456, 0.004363323129985824)
Euler angle: (0.0, -1.75, 1.8125)
Quaternion: (None, None, None, None)
Linear acceleration (m/s^2): (None, None, None)
Gravity (m/s^2): (None, None, None)

Temperature: -102 degrees C
Accelerometer (m/s^2): (-0.27, -0.3, 9.450000000000001)
Magnetometer (microteslas): (-22.0625, -29.0, -15.25)
Gyroscope (rad/sec): (0.002181661564992912, -0.004363323129985824, -0.001090830782496456)
Euler angle: (None, None, None)
Quaternion: (None, None, None, None)
Linear acceleration (m/s^2): (None, None, None)
Gravity (m/s^2): (None, None, None)

Temperature: -102 degrees C
Accelerometer (m/s^2): (-0.27, -0.29, 9.44)
Magnetometer (microteslas): (-22.75, -27.875, -14.875)
Gyroscope (rad/sec): (0.001090830782496456, 0.002181661564992912, 0.002181661564992912)
Euler angle: (None, None, None)
Quaternion: (None, None, None, None)
Linear acceleration (m/s^2): (0.02, 0.0, -0.34)
Gravity (m/s^2): (-0.3, -0.31, 9.790000000000001)

Temperature: 27 degrees C
Accelerometer (m/s^2): (-0.26, -0.3, 9.42)
Magnetometer (microteslas): (-22.375, -27.875, -14.0625)
Gyroscope (rad/sec): (-0.001090830782496456, 0.0, 0.0)
Euler angle: (0.0, -1.75, 1.8125)
Quaternion: (None, None, None, None)
Linear acceleration (m/s^2): (None, None, None)
Gravity (m/s^2): (None, None, None)

Temperature: -101 degrees C
Accelerometer (m/s^2): (-0.27, -0.32, 9.42)
Magnetometer (microteslas): (-22.375, -27.5625, -15.25)
Gyroscope (rad/sec): (-0.001090830782496456, -0.001090830782496456, 0.0)
Euler angle: (None, None, None)
Quaternion: (None, None, None, None)
Linear acceleration (m/s^2): (None, None, None)
Gravity (m/s^2): (-0.3, -0.31, 9.790000000000001)

ladyada, 2021-01-01:
please post aphoto showing your bno055

ender18g, 2021-01-03:
Thanks for the response LadyAda. Here's a photo:

[image]

ladyada, 2021-01-03:
ok please minimize the wiring to have JUST the pi and JUST the sensor and post in the adafruit forums. thanks! :)

dherrada, 2021-01-11:
@ender18g The temperature problem is a known issue and an example of how to fix it will be up pretty soon. As to some of the other values going to 'None', I have replicated this, and I think this may be due to a recent (last 2 months) change since I don't think I came across this when I was trying to fix an issue with this library in November. In the meantime, if you want to take a look at the temperature() function I added here. (https://github.com/adafruit/Adafruit_CircuitPython_BNO055/blob/temp-example-fix/examples/bno055_simpletest.py) I'm not sure if that's the final version that will be merged, but I can confirm that it works.

GuenterQ, 2021-04-18:
@dherrada Hello, I've also had this temperature issue.

My workaround is that I only translate bit0 - bit6 of the register TEMP to the temperature, and ignore bit7.

Best Regards,
GuenterQ

dherrada, 2021-04-19:
@GuenterQ Interesting. Thanks! If you'd like to make a PR go ahead!

caternuson, 2022-11-04:
Closing. Seems fixed with #79.

If still an issue, post in forums as suggested above, since it could be other things that should be investigated there.
https://forums.adafruit.com/

closed 2022-11-04 by caternuson