adafruit/Adafruit_CircuitPython_BNO055 #71: Reading time too long

aurelienbl, 2021-01-25:
Hello!

I have a quick question regarding the time it takes to read the values of the sensor.
Here is a modified version of one of the simple examples which are provided:

```

import time
from adafruit_extended_bus import ExtendedI2C as I2C
import adafruit_bno055

i2c = I2C(8)
sensor = adafruit_bno055.BNO055_I2C(i2c)

sensor._reset()    
time.sleep(0.5)
sensor.accel_bandwidth = 0x1C
time.sleep(0.5)
sensor.mode = 7 

while True:
    t = time.time()    
    print("Accelerometer (m/s^2): {}".format(sensor.acceleration))
    t2 = time.time()
    print("took {} ms".format((t2-t)*1000))
    print("")
```
   
And here the output I get:
```

Accelerometer (m/s^2): (0.0, 1.28, 9.6)
took 4.441738128662109 ms

Accelerometer (m/s^2): (-0.02, 1.24, 9.64)
took 4.462003707885742 ms

Accelerometer (m/s^2): (-0.06, 1.31, 9.61)
took 3.902912139892578 ms
```

If I understand that correctly, it means that is takes about 4-5ms to read the values of the accelerometer. And it takes the same amount of time to read the gyro and magnet values as well, so that in practice we can only have values of the sensors once every 10-15 ms.

Is there a way to use the python library and improve those reading times that prevent using high frequencies for the accelerometer/gyroscope?

Thanks in advance for the help!
Aurélien

jposada202020, 2021-05-11:
@aurelienbl, hello what would be the issue with the library?
Does not know if this is still relevant but normally we do not use Github for support questions, as they do not have the same visibility as the Adafruit Forums or the Discord channel. Thanks

caternuson, 2022-11-04:
Try without the print statements in the loop. I2C comms can also limit things, can try upping clock speed.

Closing. Not really an issue with repo.

closed 2022-11-04 by caternuson