adafruit/Adafruit_CircuitPython_BNO055 #72: Magnetometer may not be returning the correct data

dherrada, 2021-03-23:
In the datasheet, there are 6 (0x0E-0x13) registers where the magnetometer information is stored (https://cdn-shop.adafruit.com/datasheets/BST_BNO055_DS000_12.pdf, starting at page 57). However, the code is currently only using the first (0x0E) register, which should only give the lower byte of the X data. I've been playing around with fixing it for a bit, but I'm not too familiar with registers, so some guidance on what to do would be great.

tekktrik, 2021-12-03:
Just digging around at open issues, my understanding of registers is also not too deep, but it looks like the sensor auto-increments the register address in the I2C call.  That plus tracing everything down to I2C_Device, it looks good to me!

I'm less sure about how the UART version works, but my guess is that because you as pass the length as an argument, it directs the IC to return that many registers of data from the starting register.  So that would mean it's also good.

Happy to help with this if I misunderstood anything though!

caternuson, 2022-11-04:
Closing. The auto increment should take care of this. Note the format string in the register setup:
https://github.com/adafruit/Adafruit_CircuitPython_BNO055/blob/3db6570aa4e7557e0e5e79297cde34e486bf23e7/adafruit_bno055.py#L769

`h` is a short integer 2 bytes in size, so `hhh` will read 6 bytes total.

More info:
https://docs.python.org/3/library/struct.html#format-characters

closed 2022-11-04 by caternuson