adafruit/Adafruit_CircuitPython_BNO055 #79: Reading too many bits from OPR_MODE register for determing the operation mode

GuenterQ, 2021-04-18:
Hello friends,

According to Page 56 of the BNO055 datasheet (Rev. 1.7, November 2020), only bit0 - bit3 of OPR_MODE register represent the operation mode of the sensor. 

However, in file adafruit_bno055.py, line 302, all of the eight bits are returned as the operation mode, which leads to strang behaviors as reported in https://github.com/adafruit/Adafruit_CircuitPython_BNO055/issues/67. There, some physical values such as euler, quaternion etc. are shown as None, because a wrongly translated operation mode is out of range.

Suggestion:
return ( self._read_register(_MODE_REGISTER) & 0b00001111 )

Best Regards
GuenterQ

ladyada, 2021-04-18:
sure! please submit a PR :)

ViennaMike, 2021-04-21:
I concur with the issue and fix, but fyi, the Quaternion and Euler values returning NONE when they shouldn't were also sometimes caused by an error in another section of the program (issue 75, fixed by pull request 76).

GuenterQ, 2021-04-22:
This issue has been solved in https://github.com/adafruit/Adafruit_CircuitPython_BNO055/pull/80.

closed 2021-04-22 by GuenterQ