adafruit/Adafruit_CircuitPython_BNO055 #88: Number of digit reported for accelerations ang gravity components are just two

dimitrigrigoriev, 2022-01-16:
Is it possible to increase the accuracy of the reported accelerations to at least 8 digits? It is currently just two digits, however, in a 2G mode 14 bit ADC should have an LSB value of 0.002395019531250m/s^2 and we are loosing information.

ladyada, 2022-01-16:
can you be more specific about what you mean, you can print out the value iwth more digits of precision in arduino?

dimitrigrigoriev, 2022-01-16:
Hi Ladyada,
I am using Python driver on the Raspberry Pi. My output currently is:
Gravity (m/s^2): (0.62, -0.42, 9.77)
When I try to format the output with five digits it just gives extra zeros: 0.62000
It looks like the number of digits is limited when data is converted from binary to float.

ladyada, 2022-01-16:
could be - that precision may not be available!

dimitrigrigoriev, 2022-01-16:
OK, but is it possible to get raw output out in in binary or in dec? If yes than I can convert the the data into needed precision on a higher level using range and offeset data.

ladyada, 2022-01-21:
https://github.com/adafruit/Adafruit_CircuitPython_BNO055/blob/main/adafruit_bno055.py#L859
thats where the response is read from i2c, so you just have to call `bno._read_register(0x28, 6)` to get the raw data from that register

caternuson, 2022-11-04:
Closing. Looks answered.

closed 2022-11-04 by caternuson