adafruit/Adafruit_CircuitPython_BNO055 #89: How to get output in LSB for Python application?

dimitrigrigoriev, 2022-01-18:
Currently, the Python library gives an output for accelerations and g-components as floats with two-digit accuracy. Is it possible to get actual output in LSB to convert to m/s2 using range and with better accuracy?

tekktrik, 2022-03-01:
You can check the underscore attributes for the class and use the registers mentioned there to directly read the registers.  For example:

```python
class BNO055_I2C(BNO055):
    """
    Driver for the BNO055 9DOF IMU sensor via I2C.
    """

    _temperature = _ReadOnlyUnaryStruct(0x34, "b")
    _acceleration = _ScaledReadOnlyStruct(0x08, "<hhh", 1 / 100)
```

In this case, the register for temperature is `0x34` and the register for acceleration is `0x08`.  If you're familiar with how the `struct` module works, that string is the format of the return (https://docs.python.org/3/library/struct.html).

caternuson, 2022-11-04:
Closing. Above is the answer.

closed 2022-11-04 by caternuson