adafruit/Adafruit_CircuitPython_BNO055 #91: WebGL Example - Incorrect rotation on screen

alapsansky, 2022-01-25:
The motion of the 3D model in the WebGL example does not reflect the motion of the BNO055 sensor. Specifically, with the sensor on a flat surface with the VIN to RST pin holes facing away, a change in pitch of the sensor (which accurately changes the values next to "Pitch =") causes the model to roll. A change in roll causes the model to yaw. And a change in yaw causes the model to pitch. 

The system shows a complete calibration. Because the Euler output seems correct, I am guessing that there is some issue with the order of the quaternions in the code, but I have not been successful with identifying the exact issue. 

I have tested this with two sensors, using both a direct soldered connection and a QWIIC connector. 

Any guidance would be greatly appreciated!

alapsansky, 2022-02-11:
I just wanted to check in to see if anyone has experienced this same issue recently. Thank you!

hazemy, 2023-07-25:
Hello, 
I can confirm that the same issue still exists. The yaw angle motion of the sensor is reflected on the screen as pitch and vice versa. I think this is related to the BNO_AXIS_REMAP (sec. 3.4 of the datasheet). This was already part of the WebGL example of the older library (https://github.com/adafruit/Adafruit_Python_BNO055/blob/master/examples/webgl_demo/server.py#L61). Axis remaps are already added to the CircuitPython library as per #78. I could only get this fixed by using the same axis remaps used in the older library example, combined with inverting the quaternion order (W,X,Y,Z). This is a hacky fix and further digging into the mentioned section of the datasheet is necessary.