adafruit/Adafruit_CircuitPython_BNO055 #99: How to use euler as a compass heading

KingFuGitHub, 2022-03-19:
Currently the euler acts like a heading when it's turned on facing north. I wonder how I could make it so the sensor.euler[0] acts like a compass heading when not facing north originally when the IMU turned on. Thank you in advance.

tekktrik, 2022-03-21:
A brief look at the data sheet doesn't show a way to tell the sensor to compensate for the difference.  But you could always do it yourself if you know the angle!

```python
import time
import board
import adafruit_bno055

i2c = board.I2C()
sensor = adafruit_bno055.BNO055_I2C(i2c)

heading_offset = 22.1  # offset of 22.1 degrees

while True:
    print("Heading based on power on: {}".format(sensor.euler[0]))
    time.sleep(1)
```

KingFuGitHub, 2022-03-22:
That would be hard coding it because the offset isn't always the same due to the initial boot up orientation being different every time the IMU is powered on.

tekktrik, 2022-04-02:
Duplicate of #100

closed 2022-04-02 by tekktrik