adafruit/Adafruit_CircuitPython_BNO055 #101: bno055 fails to init with esp32-s3

DandrewsDev, 2022-04-03:
Can not get the bno055 library to init and not fail on a Unexpected Maker Feather-S3. 
On first run it fails with the traceback below. However any subsequent tries, soft reboots, or pushed code changes without a hard reset and it fails with "RuntimeError: No pull up found on SDA or SCL; check your wiring". So after it fails the first time the entire i2c bus stops working until a hard reset is performed. 

Traceback (most recent call last):
  File "code.py", line 9, in <module>
  File "adafruit_bno055.py", line 777, in __init__
  File "adafruit_bno055.py", line 212, in __init__
  File "adafruit_bno055.py", line 788, in _read_register
  File "adafruit_bus_device/i2c_device.py", line 140, in write_then_readinto
OSError: [Errno 116] ETIMEDOUT

Board used: https://circuitpython.org/board/unexpectedmaker_feather3/
I2c device used: https://www.adafruit.com/product/2472
Circuitpython version: Tested on 7.2.0-rc.0, 7.2.0, 7.2.4, 7.3.0-beta.0 

When just doing a simple I2c device scan, theres no issue in seeing the device.
Scanning I2C bus
0x28
1 device(s) found on I2C bus

Other I2c devices also work fine. I was able to test a pressure sensor using adafruit_dps310, on the same I2c bus without issue.

tekktrik, 2022-04-03:
Hey @DandrewsDev, would you mind posting your code as well that resulted in the error?

DandrewsDev, 2022-04-03:
`import board
import busio
import time
import adafruit_bno055

i2c = busio.I2C(board.SCL, board.SDA)
time.sleep(2)

sensor = adafruit_bno055.BNO055_I2C(i2c)

while True:
    print("Test")
    time.sleep(2)
`

tekktrik, 2022-04-03:
What immediately comes to mind is that maybe it's a hardware problem, especially if your board is able to talk with other I2C devices.  Are you able to use a multimeter and check that the pull-up resistors are working for BNO055 breakout board's SDA and SCL pinout?

tekktrik, 2022-04-03:
I wonder if you could try slowing down communication as well?  Datasheet it can do 400 kHz and `busio.I2C(...)` has a default frequency of 100 kHz, but maybe worth trying?

DandrewsDev, 2022-04-03:
I went ahead and checked the pull-up resistors and found no issue. So I grabbed another circuitpython micro I had laying around that uses an RP2040 instead of a esp32-S3. And the bno055 works great on that board with the same exact code. 

I just tied setting the frequency=50000 per your recommendation, and that works initially and then fails.

i2c = busio.I2C(board.SCL, board.SDA, frequency=50000)
time.sleep(2)

sensor = adafruit_bno055.BNO055_I2C(i2c)

while True:
    print("Test")
    print(sensor.euler)
    time.sleep(.1)
    
    
    
 After about 2 seconds it fails the same way.
 (239.0, 14.5, -152.625)
Test
(239.0, 14.5, -152.625)
Test
(239.0, 14.5, -152.625)
Test
Traceback (most recent call last):
  File "code.py", line 13, in <module>
  File "adafruit_bno055.py", line 443, in euler
  File "adafruit_bno055.py", line 147, in __get__
  File "adafruit_register/i2c_struct.py", line 39, in __get__
  File "adafruit_bus_device/i2c_device.py", line 140, in write_then_readinto
OSError: [Errno 116] ETIMEDOUT

tekktrik, 2022-04-03:
Hm, it may be a specific issue for the Feather S3.  You might want to get in touch with Adafruit Customer Support, as well as see if Unexpected Maker has any helpful advice.  In the meantime, does reducing the frequency further help at all?

DandrewsDev, 2022-04-03:
I had reached out to Unexpected Maker via their discord first, after some debugging it seemed specific to this component and library. However it appears to be a combination of the Feather-S3 and the bno055 which don't play nicely at the default frequency.

Reducing it further to 48000, and theres no issues. Ran it for a full half hour and not one hiccup or disconnect. Thank you for helping me find a work around!

tekktrik, 2022-04-04:
My pleasure, glad it's up and running smoothly now!

closed 2022-04-04 by tekktrik