adafruit/Adafruit_CircuitPython_BNO055 #109: BNO055 reset yields a bad chip id runtime error

angela804, 2022-10-10:
I'm creating a system that's supposed to utilize the ability of the BNO055 sensor to know the angle for rotation.

However, for the system to be continuous and long lasting, I need to implement a reset function where it resets the current reading of the BNO055 back to zero.

So, I created this function inside the python file under BNO055_I2C class, and around line 778 to implement the reset.

`def _resetBNO055(self):
       super()._init__()`

The system did work as it has the ability to reset the sensor reading back to 0. However, after a short while let's say for a few minutes, it shows this error. Did I miss to put something inside the reset function? Or did the delays of getting signals messed up.

`Traceback (most recent call last):
  File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Downloads/Multiprocess Copy/launcher.py", line 73, in runRover
    rover.rotate()
  File "/home/pi/Downloads/Multiprocess Copy/rover.py", line 109, in rotate
    self.stop()
  File "/home/pi/Downloads/Multiprocess Copy/rover.py", line 129, in stop
    self.headingSensor._resetBNO055()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bno055.py", line 781, in _resetBNO055
    super().__init__()
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_bno055.py", line 214, in __init__
    raise RuntimeError("bad chip id (%x != %x)" % (chip_id, _CHIP_ID))
RuntimeError: bad chip id (d0 != a0)`

tekktrik, 2022-10-11:
Are you doing any sort of threading or multiprocessing (assuming this is Blinka, but let me know if that's not true)?

angela804, 2022-10-19:
@tekktrik yes, my program's running on multriprocessing. I fixed this bad chip id using clock stretching on my raspberry pi, however, not on the OS Error Errno 121.
https://github.com/adafruit/Adafruit_Python_PureIO/issues/28

tekktrik, 2022-10-19:
Are you doing I2C read or writes in the other processes?  My guess would be that multiple I2C calls are conflicting with each other.

angela804, 2022-11-08:
@tekktrik I solved this one. An i2c clock stretching problem. I simply put this one in R-Pi's boot config
dtparam=i2c_arm_baudrate=10000

tekktrik, 2022-11-08:
That's great to hear!

closed 2022-11-08 by tekktrik