adafruit/Adafruit_CircuitPython_BNO055 #113: Typehint and docstring for quaternion reference returning a tuple of length 3 and it should be of length 4.

pjjefferies, 2023-02-18:
Typehint and docstring for quaternion reference returning a tuple of length 3 and it should be of length 4.

```py
    @property
    def quaternion(self) -> Tuple[Optional[float], Optional[float], Optional[float]]:
        """Gives the calculated orientation as a quaternion.
        Returns an empty tuple of length 3 when this property has been disabled by the current mode.
        """
        if self.mode in [0x08, 0x09, 0x0A, 0x0B, 0x0C]:
            return self._quaternion
        return (None, None, None, None)
```

jposada202020, 2023-02-19:
Thanks :). You are right. I think issue has been carry over, was introduced after https://github.com/adafruit/Adafruit_CircuitPython_BNO055/commit/ef3002498263297b5d2ae2b60ebbb743214e1ce3 in this commit https://github.com/adafruit/Adafruit_CircuitPython_BNO055/commit/f3246667f4b4d2195f19496d2d8513b1eae11b00. So we should correct both the typehing and the description. @pjjefferies would you mind do a PR, that would be very helpful .  thanks :)

pjjefferies, 2023-02-20:
Yes. I'll give it a try.

jposada202020, 2023-02-20:
Thanks :) Solved by #114

closed 2023-02-20 by jposada202020