adafruit/Adafruit_CircuitPython_BNO055 #125: Reading all zeros

researcher2312, 2024-08-21:
I have encountered a problem where the device is connected properly, but constantly reads only 0.0 results. The temperature only reads 23, no matter how hot or how cold it gets. The calibration values always read (0, 3, 0, 0). Any resetting or changing modes does not help.

dhalbert, 2024-08-21:
Which board are you using this with, and which version of CircuitPython?

researcher2312, 2024-08-22:
[image]
I am running a custom hardware with Raspberry Pi Pico W
CircuitPython 9.1.1

FoamyGuy, 2024-11-22:
I tested this today with a BNO055 QTPY bff connected to a Raspberry Pi Pico W `9.2.1` I am getting accurate values for temperature and accel / magnetometer.  I am not sure of the root cause of your issues, but I don't think it's a bug in the library. If you are still having this problem please ask for help in the forums or discord. Or if you have identified some issue within the library code related to this feel free to re-open.

closed 2024-11-22 by FoamyGuy

researcher2312, 2024-11-23:
Yes, thank you, probably my sensor unit is damaged and readings are not correct.